JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #1: Gamars PSX-001 (Hardware VCD Decoder)

realJoshByrnes, 2021-02-10:
[image]

As requested, please find attached a photo of the PCB. It identifies as [C4, CF] in UniROM.

JonathanDotCel, 2021-02-10:
Thanks @realJoshByrnes!
I've got one since we last spoke and the hardware matches - so that's good to see.

The issue right now is just that Unirom would have to monitor the switch for you to be able to use the VCD part again.
(Always boots into their custom rom and monitors the switch - it's not handled by hardware ).

If I find a foolproof way to detect this from other carts, I'll get it supported right away :)

closed 2021-02-10 by JonathanDotCel

realJoshByrnes, 2021-02-10:
Woah, thanks for the quick reply.

Awkwardly, I don't remember speaking to you earlier (sorry!) 🤭

Have you got a firmware dump of this cart?
Did you manage to flash this cart? (How?)

I personally don't use the VCD decoder (although I do have some fond memories of it in the 90s) so I'm happy to lose it if there's a working backup (if not I should be able to dump it soonish)

Ultimately it would be nice to have the ability to switch back to the hardware DVD decoding, is there anything I can do to contribute with this?

realJoshByrnes, 2021-02-10:
This device also gave me a feature suggestion... I notice when the devices ROM is booted and waiting, the CD stops spinning. I also notice when I'm running UniROM the CD spins constantly.

JonathanDotCel, 2021-02-10:
So this could totally be automatic... but it puts extra wear on the drive (especially as most machines can be unlocked).
Instead you can do ```Disc -> Disc Swap (NTSC-J)```

Sounds like that needs renamed to "Stop Disc Spinning" ?

realJoshByrnes, 2021-02-10:
I would have thought the disc stopping on boot would have put less strain on the drive than continuously running it?

What I meant was automatically stop the CD drive while it's not being used instead of constant spinning.
Eg. I load UniROM from disc, once in PSX RAM it could stop until needed. The same if I load UniROM from cart.

JonathanDotCel, 2021-02-10:
Heh nah, there's a few good reasons to leave it spinning:

1- less mechanical wear (spin up, spin down)
2- the laser has to refocus (+when changing speeds); massively more error prone than just leaving it
3- you're not waiting on the disc spinning up to start games, etc

It doesn't take much energy to keep it spinning either (vs changing speeds). Best to just pop the lid!

realJoshByrnes, 2021-02-10:
Fair enough. 👍