JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #3: Softmod alternates Castlevania SotN inputs

Robertcop84, 2021-05-14:
The softmod alternates the joypad hooking or something along those lines, even on a Japanese system when I just use it to swap the disc.
This will result in the system not being able to latch the controller correctly, which will leave players that autobooted able to use the analog stick and also makes it possible to give the game multiple directional inputs in one frame.
In the instance of SotN you can just mash left or right real fast and make it glitch out in the prolog for instance.

This behavior was prior reported on emulators and also lately on the xstation. Although even on Emulators it is the general consensus that players needed to bind the same action multiple times to hold opposite directions.

I can also tell you from prior experience, that the Innovation Super Gameconverter did that back in the day. I just did not mash on the dpad back in the day, but I remember that I always could use the analog Stick.

Robertcop84, 2021-05-14:
I just flashed the newest version to my cart. It fixed the problems, but I could not find something related to this in the changelog.

closed 2021-05-14 by Robertcop84

JonathanDotCel, 2021-05-14:
Thanks for the report Robert, glad to hear it's working.
Could you tell me though, what PSX version was this (5500 etc), which controller (including the A,M or H at the end), and which version of unirom you had previously + now.
Then at least I can keep tabs on it.
Cheers.

Robertcop84, 2021-05-14:
I have tested this on a PAL 7502 and a NTSC J 7000 with the Unirom 8.0D installed prior and changed it now to 8.0H 

The controllers I tested are 1180 A Dual Analog (N50 on the plug), 1200 H Dual Shock (with N1158 and with N50 on the plug) and a 10010 A Dual Shock for PS2  (N1158 on the plug)

JonathanDotCel, 2021-05-15:
And the double-input bug... were you able to replicate that with all of those controllers?
Did it happen without the N1158?
Thanks.

Robertcop84, 2021-05-16:
Yes with all those controllers you just needed do tap a direction a couple times (not even as fast as a double click on a mouse) or change directions rapidly.
I guess I can flash the old version again and make a better Video for you if you need.