JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #10: Motor Toon Grand Prix (JPN) freezes unless you yank the memory card out.

GoatWalker, 2022-05-19:
I already posted this on the FreePSXBoot Github page, but I figured I might as well post this here too. So I have a legit copy of the game in the title and I'm using a Memcard Pro loaded with FreePSXBoot and a PSOne (the slim one). I do have the right image on my card because I was able to easily load my copy of Fire Pro Wrestling G easily with no problems.

After FreePSXBoot does it's thing, it freezes on the Sony Computer Entertainment Presents screen. I thought something was wrong, until I pulled out the memory card and it would boot normally. So as an alternative, I used another memory card I had lying around and it was able to save and load fine and during all of that, I noticed that my Memcard wouldn't create another virtual memory card dedicated to the game. And I'm assuming the game is trying to read the memory card, but it's obviously "corrupt" according to the games standards. I have also tried this with an older Playstation (SCPH-7501) and the same thing happens unless 

TL;DR: Motor Toon Grand Prix (SCPS-10001) doesn't play well with Memcard Pro installed with FreePSXBoot, it tries to read the data on the card and create a save on said card, a VMC dedicated to the game isn't being created and because of all this mess, the game freezes after the game is loaded.

JonathanDotCel, 2022-05-22:
hiya,
are you using the newer "leave it in slot 2" images and leaving it in slot 2?
If it works for other games it might just be that this game uses different routines to try and access the card.

JackSlate6667, 2022-05-23:
No I'm not. I've been using the image for slot 1 for my Memcard Pro as recommended from FreePSXBoot's page. Like I said, the game does boot, but I need to use a different memory card because the gadget doesn't create a new VMC when it boots the game.

GoatWalker, 2022-05-23:
No I'm not. I've been using the image for slot 1 for my Memcard Pro as recommended from FreePSXBoot's page. Like I said, the game does boot, but I need to use a different memory card because the gadget doesn't create a new VMC when it boots the game.

ramapcsx2, 2022-05-23:
Hey, there's an MCPro update out today that might help. Give it a try! :)
https://twitter.com/8bitmods/status/1528804482963972102

GoatWalker, 2022-05-23:
I just updated it and no dice. So this is what happens to the Memcard when it boots. Its supposed to show the game name with the ID code or whatever its called
[image]

And this is where it freezes because on boot when it detects a new memory card it aaves automatically.
[image]

Do the the Memcard Pro people read their Twitter account or do they also happen to have a Github? Honestly, it's a minor inconvenience since I've got a couple of spare memory cards and I can swap out to those ones after boot and I can play it just fine. Its more of a convience thing since I might as well have this thing permanently plugged in.

Kind of changing topics but King's Field 1 (Jpn) also doesn't properly boot up with Unirom the usual way, but if you use the PSX.exe in the file explorer, it works fine.

GoatWalker, 2022-05-27:
So this looks to be more of Memcard Pro issue. They are going to look into it on their end.

closed 2022-05-27 by GoatWalker

Hugeack, 2022-08-01:
Hey, same issue with King's Field (SLPS-00017). I should mention that it launches fine on my SCPH-5501 in `8.0.I`ish but stalls in `8.0.J` if you don't launch the executable manually. Don't know if this regression is fixed in the upcoming version, but it sounds probable. If not, I guess I'll open a separate issue when the time comes?