JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #11: Gamars Movie Cart ROM

Annunaki, 2022-05-22:
Hi Guys,

I have a Gamars Movie Cart which has a PAL v1.01 ROM already flashed on it, I notice in the ROM collection on Unirom that there is a Gamars PAL v2.00 ROM available however when selecting "auto-detect cart type" it returns a message saying "can't identify the cart or eeprom chip" and also prints out an eeprom ID of CC, 03.

I also notice that in the Gamars ROM list that ROM version of PAL 1.01 has not yet been archived but the NTSC version has, Could this be the cause of the above error? How can I go about dumping this cart's ROM, so that we can get it archived for future use and all that.

I apologize if this is not the place for this question, but I am new to this scene.

JonathanDotCel, 2022-05-22:
Ooh, nice thanks for reaching out.
It's easy enough to dump without identifying the ROM (just awkward to write), so maybe nobody's got around to it yet?

Are you able to jump on discord?
https://discord.gg/QByKPpH or http://psx.dev

The ROM list on the cart isn't complete (for the sake of disc size), but Squaresoft74 on discord (and kHn I think) are archiving a pretty complete list of ROMs and would know best.

Could run you through setting up a serial connection, or if you have some way of writing/reading to memcards, maybe patch something in to do that?

Annunaki, 2022-05-25:
Hey, sorry I didn't come back here, Just finding my bearings with all the different resources in the scene! I think we just spoke on discord, so if that was you, then yeah, I will wait for you to get back to me! Cheers.

JonathanDotCel, 2022-05-27:
Hah, yeah that was me, I'll give you a nudge!

JonathanDotCel, 2022-09-10:
8.0.K (releasing) today allows dumping to memcard, thanks :)

closed 2022-09-10 by JonathanDotCel

Annunaki, 2022-09-11:
Hi,

I tried to dump this cart's ROM to memory card, however, receive the message:

CHECKING CART...
DETECTED EEPROM ID: CC, 3
UNKNOWN EEPROM! (CC,03)

COULDN'T IDENTIFY THE CART OR EEPROM CHIP!

IT MAY BE UNSUPPORTED OR INCORRECTLY INSERTED!