JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #13: Open Source The ROM Builder (But Not Neccesarily The UNIROM Source)?

alex-free, 2022-06-21:
I am highly interested in running my own PS-EXE on a cheat cartridge, specifically my modified Tonyhax loader found at https://github.com/alex-free/tonyhax .  Your build method for the ROM image is exactly what I want to do with the loader. I know the source is not ready for release, but what about just the methods to build a PS-EXE into a ROM image that runs it?

The tonyhax loader fork allows for playing backups with correct CD audio  on all Japanese consoles, something no other backup method or tool can do.

c0deisme, 2022-06-23:
https://github.com/JonathanDotCel/RomProd
this seems to do the job aswell

JonathanDotCel, 2022-06-23:
> 

hiya, yeah the link c0deisme left should work just fine for your purposes. gimme a shout if you need anything.

"The tonyhax loader fork allows for playing backups with correct CD audio on all Japanese consoles, something no other backup method or tool can do."
Wondering about this though, I had a wee peek through the source - do you mean the TOCPerfect (disc swap), 'cause unirom already does this.
Cheers.

JonathanDotCel, 2022-06-23:
Lmao, just spotted "void SetSessionSuperUltraCommandSmash()	" This looks fun!

ramapcsx2, 2022-06-23:
Chiming in as well :p 

SetSessionSuperUltraCommandSmash(): 
I noticed SetSession not updating the Mechacon internal TOC on some older machines. Is this hammering the command until the TOC eventually does update?

alex-free, 2022-06-23:
> Chiming in as well :p
> 
> SetSessionSuperUltraCommandSmash(): I noticed SetSession not updating the Mechacon internal TOC on some older machines. Is this hammering the command until the TOC eventually does update?

In simplest terms yes. Early 1000/1001/1002s need to be hammered to update the TOC.

I wrote a C program to write PS-EXEs into a gameshark ROM and am now just seeing this haha.

JonathanDotCel, 2022-09-10:
Heh cheers @alex-free!
I haven't had time to experiment with ultrasupermegasmash() for *this* release, but your rom's on the bootdisk now
(I believe squaresoft74 has sorted all of this out :D)

Plan's to opensource shortly, so I'll close this for now

closed 2022-09-10 by JonathanDotCel

alex-free, 2022-09-11:
@JonathanDotCel me and MottZilla actually figured out a much faster method abusing the timing of reading a non-existant Session 2 on a non-mulit-session disc which works similarly but much faster and is much easier to implement! This new method is in that ROM and Tonyhax International's source if you want to check it out. It is very easy to use.

SetSession 2 (switch to session 2)
wait for 2 interrupts (there is actually 3 because the second session does not exist but we ignore that to make this work :))
immediately send SetSession 1 (switch back to session 1)
Complete TOC data is read without loss of authentication!

Again this only affects all SCPH-1000 and very early SCPH-3000 units which all have either VC0 A or VC0 B CDROM BIOS Controller firmware. 

Thanks for including the ROM!