JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #14: JP Legend Of Mana

L10N37, 2022-07-17:
JP Legend of Mana is anti mod flagged.

JonathanDotCel, 2022-09-10:
Hmm, I've had a play around, but there's not a ton I can do about this...
Disc swap, mod chip and mecha unlock all trigger protection.

Are patches avaialble?

L10N37, 2022-09-11:
I'm more interested in how It works as I can't read Japanese :P

I am keen to see if Xstation / PSNEE trip it but I am not set up with either ATM (moving house).

Basically (for me anyway) just a curiosity as I have a modchip I've got several revisions of.

ramapcsx2, 2022-09-11:
The game works fine with Unirom + xStation.
I couldn't test Unirom + CD-R, but it is generally a normal modchip protected title. It does the manual check for one, as do many other games.

L10N37, 2022-09-11:
There's definitely something different about the check on this particular title. The laser isn't positioned down towards the smallest circumference on the disc when it gets flagged either. I can get past it sometimes on a PU23 but not a PU22 with no changes to how the chip works.

ramapcsx2, 2022-09-11:
The game first calls ReadToc, which is the common check that the shell also does. The laser gets placed to the lead-in area and symbols arrive.
Then the game does the extra check that is common to modchip protected titles: It seeks out towards somewhere in the middle of the disk, and listens for symbols again. If they arrive, a modchip got detected.

L10N37, 2022-09-12:
ahh so it's just the timing of the checks that differs greatly to other titles with similar checks