JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #17: Dino Crisis (NTSC-U/C) - "Software terminated"

Keiyuu, 2022-09-30:
I'm getting the infamous "software terminated" anti-modchip message on Dino Crisis (US Greatest Hits version, SLUS-00922GH), burned on a CD-R. I'm booting Unirom through FreePSXBoot on a Memcard PRO.

I've read that tonyhax patches these games automatically to bypass this. Is there way to do the same on Unirom or are these games simply unplayable though Unirom?

ramapcsx2, 2022-10-02:
A clean image will run a modchip detection routine, which will have to be patched (as you guessed).
Unirom doesn't do these patches, so you'll have to find a way around it. You could patch the game, or find a way to run Tonyhax.

Another alternative is a good old modchip :p

Keiyuu, 2022-10-02:
> Unirom doesn't do these patches, so you'll have to find a way around it. You could patch the game, or find a way to run Tonyhax.

I'm not an expert on this but aren't Unirom and Tonyhax two similar tools with the same purpose? Wouldn't it be possible to implement the same patching routine Tonyhax is doing into Unirom, to bypass the detection routine?

I've tried booting it from Tonyhax and it does indeed work but I would prefer to launch all my games through Unirom because it has better Memcard Pro support and is automatically recognizing the game ID and creating a game-specific VMC on the Memcard. This is something Tonyhax doesn't do.

ramapcsx2, 2022-10-02:
@JonathanDotCel ? :p

JonathanDotCel, 2022-10-03:
> <JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1/issues/17/1264743400
> @github.com>

Hah, I'll look into adding it.

alex-free, 2022-11-20:
The Tonyhax International method of bypassing this would be quite easy to implement for you @JonathanDotCel 
All of the fun stuff is here:
https://github.com/alex-free/tonyhax/blob/master/loader/patcher.c

alex-free, 2022-11-20:
Also, for many games they can be directly patched and 'just work' after being burned using https://alex-free.github.io/aprip

JonathanDotCel, 2022-11-20:
> Also, for many games they can be directly patched and 'just work' after being burned using https://alex-free.github.io/aprip

Hi Alex, thanks for getting in touch.
I had a look at your code, but didn't feel comfortable using it until I'd open sourced Unirom - would feel a bit like stealing!. (hopefully by the end of the year)

Cheers for the links :)

alex-free, 2022-12-10:
> Hi Alex, thanks for getting in touch. I had a look at your code, but didn't feel comfortable using it until I'd open sourced Unirom - would feel a bit like stealing!. (hopefully by the end of the [...]

My anti-piracy bypass system has been upgraded (and is now in a different file which has broken the previous link to my code). Here is the new code link:  https://github.com/alex-free/tonyhax/blob/master/loader/ap-bypass.c . Much more optimized then before, with way more games supported. I look forward to open source UNiROM, that was part of why I wanted to make tonyhax international. Because UniROM is closed source.

JonathanDotCel, 2022-12-13:
Aww, I've just been giving out the source since I haven't had much time to clean it up, but no worries, the scene's better of for having more stuff :D
Ty for the links, I really look forward to reading through!

Keiyuu, 2024-01-07:
Any news on this?