JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #18: Availability of raw UNIROM binaries to use with EPROM programmers

hkzlab, 2022-10-26:
Hello,
I have a few carts with EPROM chips in them (so not directly re-flashable from the PSX) that I socketed.

In the past I flashed UNIROM on an xplorer cart with an EEPROM, read the flashed chip, then erased/reprogrammed the chips on my other carts, but this is quite cumbersome to do everytime I need to update.
I could also replace the ROMs inside the cards with EEPROMs, but I have a big stock of 27C010 I'd like to keep using.

Long story short: any hope of getting a plain binary file I can flash on my carts directly included in the release? Or at least instructions on how to convert from the ROM file included in the disc to the version that actually gets flashed (as I found the flashed file I read back from the EPROM different from what was included in the disc)? Doesn't matter if it's just a description of the conversion process, I can code myself a tool if that is what it's needed.

Thanks

hkzlab, 2022-10-26:
Scratch all of the above. I noticed the ROMs I dumped from the flash of the carts I programmed are different just because the programming does not erase the whole EEPROM, so residuals of the previous program are left. I can just use the provided images!

closed 2022-10-26 by hkzlab

JonathanDotCel, 2022-10-26:
Haha, yep, just to confirm, it's the raw .rom that goes on the cart.
(There are "eeprom wipe" images on the CD if you want to zero them to make diffs easier, etc btw)

hkzlab, 2022-10-27:
Yep, thanks :)
Realized what was happening when I spotted a Tekken 3 cheat reference in my dumps!