JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #19: SCPH-1000 & Early SCPH-3000 Require Special SetSession Sequence Due To Buggy CDROM Controller Firmware

alex-free, 2022-11-20:
SCPH-1000 and early SCPH-3000 consoles contain either VC0 A or VC0 B CDROM controllers. When you send SetSession on these controllers the TOC is not updated due to a bug (confirmed by NO $ PSX in an email). Tonyhax International is the only program that has a work around for this currently AFAIK. Me and MottZilla came up with it, we dub it the 'SetSessionSuperUltraCommandSmash'. Basically how it works is this:

- SetSession is sent to change the session to session #2 (which doesn't have to exist btw, and normally does not).
- We wait 2 interrupts, and then IMMEDIATELY send SetSession to change the session back to session #1 . This forces the TOC to update. There is actually a 3rd interrupt that can trigger when session 2 does not exist, we are ignoring this (undefined behavior) so that the CD drive doesn't actually lock up. 

Probably a really good idea to disable interrupts and then re-enable them after doing this, I do this in tonyhax international by calling the 'bios reinitialize' function (which does other things to, but they are probably not necessary for i.e. unirom).

Let me know if you have questions. Check out the Tonyhax International implementation here:
https://github.com/alex-free/tonyhax/blob/master/loader/secondary.c

ramapcsx2, 2022-11-20:
Thanks for documenting your findings, really appreciate that! :)

JonathanDotCel, 2022-11-20:
Awesome, once again, thanks for sharing :)

JonathanDotCel, 2023-05-07:
Thanks!
So the c++ conversion and opensourcing is well underway right now.
Just wondering if you're on discord?
Cheers,
J

alex-free, 2023-05-07:
> Thanks!
> 
> So the c++ conversion and opensourcing is well underway right now.
> 
> Just wondering if you're on discord?
> 
> Cheers,
> 
> J

I might make an account for discord.

That's awesome that it's finally happening. I look forward to contributing the SetSessionSuperUltraCommandSmash and anti-piracy bypass system. I only know a bit of C though, is that going to be allowed for UniRom with it becoming C++?

If you are also interested, I just finished implementing GameShark code support in Tonyhax International. There is a desktop program that generates a save file from a .txt full of GameShark codes. That save file can then be copied to a PSX memory card and then it is read by Tonyhax International and it applies the cheats before booting a game. I could implement something similar for UniRom as well if you are interested.

JonathanDotCel, 2023-05-08:
That's awesome man, would be nice to see you there; great little dev community.

When I say C++, it's "C, but with C++ namespaces, and the odd ENUM" lol.
Nothing mental.

That's awesome about the GS support!
Nic Noble's written a cool little cheat engine which turns the codes into machine code for maximum fast. It's not implemented yet, but I'm itching to use it!

Hope to see you soon anyway ^^

JonathanDotCel, 2023-05-17:
Haah, that's fuckin sweet - very concise!

So the one @nicolasnoble wrote generates CPU instructions for the codes, lol. E.g. trading a bit of space for speed

alex-free, 2023-05-17:
> Haah, that's fuckin sweet - very concise!
> 
> 
> 
> So the one @nicolasnoble wrote generates CPU instructions for the codes, lol. E.g. trading a bit of space for speed

That's what I was hoping you meant. That's so interesting! I don't think you can optimize more then that.