JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #20: UniROM 8.0.K cd hangs when ran on duckstation.

kotenok2000, 2022-12-24:
UniROM 8.0.K cd hangs on "Checking for settings..." when ran on duckstation.
8.0.J doesn't hang.

ramapcsx2, 2022-12-28:
Hey, there should be something like a "developer console" or "terminal", with something like "show TTY" option in duckstation.
If you enable that, you should be able to see what Unirom prints. Is it hanging on "Drive still seeking" by chance? :)

kotenok2000, 2022-12-28:
[  143.2237] E(ExecuteTestCommand): Unknown test command 0x60, 2 parameters
[  144.3570] W(LoadDataFIFO): Attempting to load empty sector buffer
[  144.3902] W(ProcessDataSector): Interrupt not processed in time, 
missed 1 sectors
[  144.4072] W(ProcessDataSector): Interrupt not processed in time, 
missed 2 sectors
[  144.4239] W(ProcessDataSector): Interrupt not processed in time, 
missed 3 sectors
[  144.4406] W(ProcessDataSector): Interrupt not processed in time, 
missed 4 sectors
[  144.4411] W(ProcessDataSector): Interrupt not processed in time, 
missed 5 sectors
[  144.4574] W(ProcessDataSector): Interrupt not processed in time, 
missed 6 sectors

kotenok2000, 2022-12-28:
Also

  [  125.4785] I/Bus: TTY: PS-X Realtime Kernel Ver.2.5
[  125.4795] I/Bus: TTY: Copyright 1993,1994 (C) Sony Computer 
Entertainment Inc.
[  125.4905] I/Bus: TTY: KERNEL SETUP!
[  125.4932] I/Bus: TTY: Configuration : EvCB    0x10 TCB     0x04
[  125.6434] I/Bus: TTY: System ROM Version 2.2 12/04/95 A
[  125.6441] I/Bus: TTY: Copyright 1993,1994,1995 (C) Sony Computer 
Entertainment Inc.
[  126.9308] I/Bus: TTY: ResetCallback: _96_remove ..
[  131.8184] I/Bus: TTY: System Controller ROM Version 95/05/16 c1
[  135.0627] I/Bus: TTY: SetGraphDebug:level:1,type:0 reverse:0
[  139.0444] I/Bus: TTY: BOOTSTRAP LOADER Type C Ver 2.1 03-JUL-1994
[  139.0595] I/Bus: TTY: Copyright 1993,1994 (C) Sony Computer 
Entertainment Inc.
[  139.1778] I/Bus: TTY: setup file    : cdrom:SYSTEM.CNF;1
[  139.3120] I/Bus: TTY: TCB     00000004
[  139.3124] I/Bus: TTY: EVENT   00000010
[  139.3129] I/Bus: TTY: STACK   801ffff0
[  139.3139] I/Bus: TTY: BOOT =  cdrom:\UNIROM_B.EXE;1
[  139.3272] I/Bus: TTY: argument =
[  139.3275] I/Bus: TTY: KERNEL SETUP!
[  139.3284] I/Bus: TTY: Configuration : EvCB    0x10 TCB     0x04
[  139.3301] I/Bus: TTY: boot file     : cdrom:\UNIROM_B.EXE;1
[  140.0966] I/Bus: TTY: EXEC:PC0(801b0000)  T_ADDR(801b0000) 
T_SIZE(00021000)
[  140.0970] I/Bus: TTY: boot address  : 801b0000 801ffff0
[  140.1131] I/Bus: TTY: Execute !
[  140.1137] I/Bus: TTY:                 S_ADDR(801ffff0) S_SIZE()

ramapcsx2, 2022-12-28:
Cheers :)
That is a curious message. I think the crew is going to have to check it out together.. It's probably not the test command (0x60), but maybe it's related..