JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #21: Mild compatibility issue with a certain game's mode.

Mike77GH, 2023-02-06:
Tested on PlayStation PAL SCPH-1002 with FreePSXBoot.
Game tested: Vib-ribbon (PAL)
The game functions normally in the tutorial and included courses, however, in the "play with my own choice of music CD" side of the game, something is breaking the function of this mode. When track 1 is selected, the drive skips ahead to track 2 instead.

JonathanDotCel, 2023-05-07:
Hi, does this only happen when using the auto disc unlock?
E.g. do you know if it also happens when using the swap trick, or a modchip, etc?
Thanks.

Mike77GH, 2023-05-08:
Yes, this happens with the auto disc unlock. As far as I know, the swap trick method has no such problem with vib-ribbon's "play with my own music CD" mode. Unfortunately I do not have a modchip on my console's, nor do I plan to install one.

JonathanDotCel, 2023-07-24:
Oki, I might have a fix for this, but it could be a short while (finishing opensourcing things).
Thanks for letting me know!