JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #22: Unirom 8.0k w/ Caetla will not apply GameShark Codes (settings will not save either 256-512k ROM cart)

JMaxxL, 2023-03-04:
Caetla bundled with 8.0k or 8.0J will not save gameshark codes or apply after booting. No settings codes or options save when useing cheat cart tried multiple 512k rom carts.

8.0K Caetla is not in english (please change language back to english or add option to select which Caetla version to use)
"Switch to Caetla 0.38 by default"

8.0J is in english but has same issue were gameshark codes or settings to not save or apply when installing UniROM w/Caetla

JonathanDotCel, 2023-03-04:
Hi,
yep, this version of caetla uses the memory cards by default - which is not ideal. Next version will be back as it was. Cheers.

closed 2023-03-04 by JonathanDotCel

JMaxxL, 2023-03-04:
ok because i keep tonyhax in slot two is this causing my issue possible, Im pretty sure I did boot up atleast once without it in slot 2

JMaxxL, 2023-03-04:
O a "gameshark memory card" Im sorry... Ive looked for quite a while I cant find any gameshark code memory card images .MCD

JMaxxL, 2023-03-04:
yeah I've checked PSX.dev for sure for gameshark memcard images. I'm happy to just have flashcart for most games but I figured if work had been done to get caetla to work id like to use it. ive used CDX GS/AR CDs they hand right after booting the game 

Thanks ! for all the work I use UniROM almost daily on my cheatcarts is there a discord or donation link anywhere

JMaxxL, 2023-03-05:
I was going to order a gameshark memory card looks like the actin replay version is cheaper will either of them work with caetla? looks like all they are is a normal memory card with a un-deleteable files right?

JonathanDotCel, 2023-03-05:
Heh, I'm actually not really sure if there are any good collections of memcard cheats, etc!
No donation link, but give something to a charity if you like.

https://www.psx.dev/ has a discord link, maybe someone will have more info about the caetla cheats?