JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #23: 8.0k on parallel Cheat Cart "Please reseat cart and try again error"

JMaxxL, 2023-03-04:
It's unclear what is different about the UniROM auto install vs Cheat Cart restore. But I was able to get a corrupted cart to work this way and another that would not even boot into its normal screen to work this way

If you have a unirom cheat cart ask you to re seat and try again, try a cart restore under install first. This will allow UniROM to be installed on 2nd boot. 

might be a good idea to do a format 128/256/512k in between restore and install my corrupted cheat cart required this

JonathanDotCel, 2023-05-07:
Hey, did you get to the bottom of this?
Sounds like it since we chatted on discord, J.

JMaxxL, 2023-07-22:
So my cart is getting extremely hot especially when i access the V-mem but it is working now so weird I flashed it many many times with alot different roms. possible 30 by now & the OEM 3.2 is working on it now. But I still worry there is a micro crack or something because it gets so dam hot.