JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #24: Support for memory that are still in production, SST 39SF040

sirTentacle, 2023-03-19:
There already are support from SST 39SF020, please add support for SST 39SF040.
Or ability to force flash an undetected IC

JMaxxL, 2023-05-03:
Sorry new to the cellphone app think I'm sending messages to wrong people but I'm having issues dose SCPH 7501 DO swop tricks ?

JonathanDotCel, 2023-05-07:
TBH, it's probs already supported, but just not in the list.

Forcing a flash would only get you so far since there's a bunch of variations on the protocol, some requiring an erase, some requiring unlocks, some using slightly different sequences, etc.

But if you send me the ID byte of this chip from the status screen, I can add it and send you a binary/rom to test.
(nudge me on discord? psx.dev)

Already supports
39SF020 & 39VF040, so I imagine it's very similar.