JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #25: Found a cart that does not work "POWER REPLAY"

loibucket, 2023-03-21:
I used the 8.0.K boot disc on a modded ps1.  I get the "ERROR FLASHING!" message as soon as I hit X.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

JonathanDotCel, 2023-03-22:
Thanks, that's some fantastic documentation!
Could you help with one or two things though please:

1: with the default install, are you able to save settings?
E.g. L1/R1 + Select, etc
If you set it to e.g. japanese or PAL/NTSC, does it store the settings?

2: How far does it get before "Error flashing"?
E.g. does it just fail immediately?

Cheers

loibucket, 2023-03-22:
1 - no it did not store settings, a dialogue asking to switch to PAL pops up, but nothing changed when I selected YES.
2 - it fails immediately 

Looks like it's write protected?  This doc talks about the Software-protected Data Write
https://www.datasheet.live/pdfviewer?url=https%3A%2F%2Fpdf.datasheet.live%2Fc1fd2ef5%2Fwinbond.com%2FW29C011A-15.pdf

JonathanDotCel, 2023-05-07:
Do you have the ID bytes from the status screen by chance?
We've already got unlock codes for a bunch of chips, it's probably already implemented but just not for this chip.
J

loibucket, 2023-07-22:
What is the ID bytes, how do I find it?

johnbaumann, 2024-03-17:
[image]