JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #30: Support for Pro Action Replay 2 (Japanese) with 2x128K EEPROMs

p-b-p, 2023-05-05:
Pro Action Replay 2 (Japanese) EEPROM is recognized as 128K only. Manual Datel V2 flash does not work.

[image]
[image]
[image]
[image]

JMaxxL, 2023-05-05:
I would do manuel generic if automatic didn't work, actually NXflash
sometimes can get around problems to. Eventually an open-source version of
NXflash may be released. This will actually widen the compatibility quite a
bit but that is rumored as a future project only :-(

p-b-p, 2023-05-05:
The manual 128k generic or automatic flash works, but it flashes the standalone version of the UniROM (128k only the first EEPROM bank).
The unirom_withcaetla.rom is bigger 265k and will not fit as only the 1st bank is of the dual EEPROM is reconized.

JMaxxL, 2023-05-05:
> <JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1/issues/30/1536149527
> @github.com>

You actually can install CAETLA alone with NXFlash version 34 allows
setting to be saved on the card. not sure if it boots homebrew BUT you
could use memcard for unirom & cart for CAETLA

I'll get the link when I get home

p-b-p, 2023-05-05:
CAETLA is not a problem I can flash it as well, but would like to have both of them on the cartige. Tested with 37 and 38 as there are only those two version on the UniROM 8.0.K CD image.

JonathanDotCel, 2023-05-07:
@p-b-p Did you try 256k generic?
if that doesn't work you can
- send me the cart to test
- tell me where you got that one

J

p-b-p, 2023-05-10:
@JonathanDotCel I've tried that option as well but it doesn't work :(

- Where I need to send it in order to test?
- I've got it from ebay UK as an imported item from Japan.

JonathanDotCel, 2023-07-24:
@p-b-p Can you PM me or come ping me on psx.dev's discord?
I'll cover postage and send it back. Thanks :)

p-b-p, 2023-08-21:
@JonathanDotCel I've sent you PM on psx.dev's discord.

petar_70881