JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #36: Unable to browse files on CD

jeffbrl, 2024-04-28:
My objective is to write freepsxboot to a memory card for use on a slim NTSC-J PSX. The PSX running Unirom is an unmodified NTSC-U SCH-5501. I burned psxboot images to a Verbatim CD. When I attempt to manually select a file, I get an error: freepsxboot cannot be found. I tried to use File Manager to browse the contents of the CD. The software seems to think the CD is blank. I have read the wiki but I remain unclear about what unicom needs to read a CD.  May I please get guidance for how to proceed?

Btw, thank you for writing this software and sharing it with the community.

ramapcsx2, 2024-04-30:
Hey, maybe I can help a bit :)
The image you burn to CD has to include all the roms in their expected folders for the tools.
Is that "psxboot images" correct, or do you rather just need the regular Unirom bootdisk release?
https://github.com/JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1/releases/tag/8.0.K

jeffbrl, 2024-04-30:
Hey Rama (love the GBSC, btw). I haven't seen any documentation on what folder structure Unirom expects. I extracted https://github.com/brad-lin/FreePSXBoot/releases/tag/v2.1 to a folder called "psxboot". There are two subfolders: "slot1" and "slot2". I apologize...I do not understand you question about the Unirom bootdisk release. I launch Unirom via freepsxboot.

jeffbrl, 2024-04-30:
I will need to explain my understanding of your comment to ensure that I am correctly interpreting. I gather that you are suggesting that I burn a different version of Unirom than the one bunded with freepsxboot. Boot using swap trick or parallel port device. This unirom version will give me the ability to write various versions of freepsxboot as they are included in the ISO image you linked.

ramapcsx2, 2024-04-30:
Correct :p

jeffbrl, 2024-04-30:
Rama's suggestion worked on the first try. :)

I suspect the unirom version bundled with freepsxboot must not have the freepsxboot MCD images included.

For those coming across this issue, here is the sequence I recommend for using a phat PSX to write freepsxboot for a slim.

1. Obtain memory card with freepsxboot for phat on ebay/aliexpress or use modded PS2 to write it
2. Burn the complete unirom version to CD
3. Launch unirom via freepsxboot exploit
4. Insert unirom disc and select Boot CD in the unirom menu
5. Use the complete unirom version to burn the MCD image you need

No messy swap trick needed.

closed 2024-04-30 by jeffbrl