JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #37: List of supported EEPROMs ?

ABelliqueux, 2024-05-25:
Hi there :)

Would it be possible to get a list of supported EEPROMs ?

I gather from the issues and release notes that right now, at least the following are supported :  

AMD AM29F010
AMD AM29F020
Sanyo LE28C1001
SST 39SF020
SST 39SF040 (pending, waiting for chip's ID byte )
SST 39VF040
SST 28SF040

Is that correct ?

Thanks !

JonathanDotCel, 2024-05-25:
Hey, here's the list from the cart detector :)

`    // DIP & PLCC package tested by retrofitting to an AR cart
    {Brand::AMD_ID_01, Chips::AMD_AM29F010, SIZE_128K, 128, (Special::ERASEFIRST | Special::PROGRAM_EVERY_BYTE), SubType::NONE, "AMD AM29F010",
    "GameHunter, etc"},

    {Brand::ATM_ID_1F, Chips::ATM_D5_29C010A, SIZE_128K, 128, Special::NONE, SubType::NONE, "ATMEL 29C010A", "Action Replay, etc"}, 
    {Brand::ATM_ID_1F, Chips::ATM_35_29LV010A, SIZE_128K, 128, Special::NONE, SubType::NONE, "ATMEL 29LV010A", ""}, 
    {Brand::ATM_ID_1F, Chips::ATM_DA_29C020, SIZE_256K, 256, Special::NONE, SubType::NONE, "ATMEL 29C020", ""}, 
    {Brand::ATM_ID_1F, Chips::ATM_BA_29BV020, SIZE_256K, 256, Special::NONE, SubType::NONE, "ATMEL 29BV020", ""}, 
    {Brand::ATM_ID_1F, Chips::ATM_A4_29C040A, SIZE_512K, 256, Special::NONE, SubType::NONE, "ATMEL 29C040A","Xplorer FX 2"}, 
    {Brand::ATM_ID_1F, Chips::ATM_C4_29xV040A, SIZE_512K, 256, Special::NONE, SubType::NONE, "ATMEL 29xV040A", ""},  // Doesn't work with the .50 PRO German FX ROM

    {Brand::SST_ID_BF, Chips::SST_07_29EE010, SIZE_128K, 128, Special::NONE, SubType::NONE, "SST 29EE010", "Smart Cartridge?"}, 
    {Brand::SST_ID_BF, Chips::SST_08_29xE010, SIZE_128K, 128, Special::NONE, SubType::NONE, "SST 29xE010", ""}, 
    {Brand::SST_ID_BF, Chips::SST_22_29EE010A, SIZE_128K, 128, Special::NONE, SubType::NONE, "SST 29EE010A", "",},
    {Brand::SST_ID_BF, Chips::SST_23_29xE010A, SIZE_128K, 128, Special::NONE, SubType::NONE, "SST 29xE010A", ""},

    // 1st Gen 1-chip Xplorer
    // 2-chip xplorers
    // 1-chip Datel V2 (Equalizer, Gameshark, Action Replay)
    // 1-chip GameBuster (Clone)
    {Brand::SST_ID_BF, Chips::SST_10_29EE020, SIZE_256K, 128, Special::NONE, SubType::NONE, "SST 29EE020", "DatelV2, XPGen1, 2Chip XP"},
    {Brand::SST_ID_BF, Chips::SST_12_29xE020, SIZE_256K, 128, Special::NONE, SubType::NONE, "SST 29xE020", "DatelV2, XPGen1, 2Chip XP"},
    {Brand::SST_ID_BF, Chips::SST_24_29EE020A, SIZE_256K, 128, Special::NONE, SubType::NONE, "SST 29EE020A", ""}, 
    {Brand::SST_ID_BF, Chips::SST_25_29xE020A, SIZE_256K, 128, Special::NONE, SubType::NONE, "SST 2xEE020A", ""}, 
    {Brand::SST_ID_BF, Chips::SST_04_28SF040, SIZE_512K, 256, Special::ARPRO, SubType::NONE, "SST 28SF040", "AR/GS Pro"},

    // Did this chip ever appear in Datel products?
    {Brand::WIN_ID_DA, Chips::WIN_C1_29EE01x, SIZE_128K, 128, Special::NONE, SubType::NONE, "WINBOND 29EE01x", "Pswd Cart, PSKey"},

    {Brand::WIN_ID_DA, Chips::WIN_45_29C020, SIZE_256K, 128, Special::NONE, SubType::NONE, "WINBOND 29C020", "unknown"},
    {Brand::WIN_ID_DA, Chips::WIN_46_29C040, SIZE_512K, 256, Special::FX, SubType::NONE, "WINBOND 29C040", "Xplorer FX1"},

    // Spotted 2 versions of this chip:
    // WORKING version in a Game Hunter with ID "D0,01" which was actually an SST NHJ29EE010
    // NONWORKING Intel/CSI 28f010 requires 12v to program, factory new you cannot save settings, etc.
    {Brand::CSI_ID_D0, Chips::CSI_01_CAT28F010N, SIZE_128K, 128, Special::NONE, SubType::NONE, "CSI cat28f010n", "GameHunter/Pswd"},
    {Brand::SANYO_ID_D0, Chips::SANYO_04_LE28C1001, SIZE_128K, 128, Special::NONE, SubType::NONE, "SANYO LE28C1001", "Pro Power Replay?"},  // BF,07 TODO! // and 60 03

    //
    // Homebrew entries
    //
    // TODO: schnappy may have a the 29sf020 with split ROM
    // EDIT: square has one of thsese in an Equalizer V3!
    {Brand::SST_ID_BF, Chips::SST_BF_39SF020, SIZE_256K, 256, Special::ERASEFIRST | Special::PROGRAM_EVERY_BYTE, SubType::NONE, "SST 39SF020", "Datel V3"}, 
    {Brand::SST_ID_BF, Chips::SST_BF_39VF040, SIZE_512K, 256, Special::ERASEFIRST | Special::PROGRAM_EVERY_BYTE, SubType::NONE, "SST_BF_39VF040", "homebrew?"},`

ABelliqueux, 2024-05-25:
Great, thanks ! Here is a nice table : 

| Brand | Model
| --- | --- |
| AMD |  AM29F010 |
| ATMEL | 29C010A |
| ATMEL | 29LV010A |
| ATMEL | 29C020  |
| ATMEL | 29BV020 |
| ATMEL | 29C040A |
| ATMEL | 29xV040A |
| SST | 28SF040 |
| SST | 29EE010 |
| SST | 29xE010 |
| SST | 29EE010A  |
| SST | 29xE010A  |
| SST | 29EE020 |
| SST | 29xE020 |
| SST | 29EE020A  |
| SST | 2xEE020A |
| SST | 39SF020 |
| SST | 39VF040 |
| WINBOND | 29EE01x |
| WINBOND | 29C020 |
| WINBOND | 29C040 |
| CSI | CAT28F010N |
| SANYO | LE28C1001 |

Not sure about SST NHJ29EE010 and CSI CAT28F010N in the end (regarding your comments ) ? Are they flashable ?

Apart from them, every model in here should be flashable from a PSX cart right ? 

Another question (sorry!) about that "ERASEFIRST/PROGRAM_EVERY_BYTE" flags; I'm using the PSX+cart+nops as an EEPROM programmer right now (because :sunglasses: ) - flashing Master System 128KB and 256KB roms to SST 39SF020 to be specific - and it works ( the newly flashed EEPROM run fine in a Master System console ).

I'm wondering if that method would work with a chip that doesn't have these flags  (using a 29C010A for example) ?

Also, I'm planning on testing using a 39SF040 soon© so I should be able to give you the ID bytes for this chip.

danhans42, 2024-05-25:
The CAT28F010 needs +12v VPP so won't be flashable without a hardware modification to provide that unfortunately 🙂

https://www.onsemi.com/pdf/datasheet/cat28f010-d.pdf

JonathanDotCel, 2024-05-25:
> Great, thanks ! Here is a nice table :
> 
Neat. thanks!

> Incompatible chips (need 12V) :
> 
> Brand	Model
> CSI	CAT28F010N
> SST	M28F101 (reported as CAT28F010N)

That's right, yea

> ~Not sure about SST NHJ29EE010 in the end (regarding your comments ) ? Are they flashable ?~

So it has the wrong ID, but it does flash properly!

> Apart from those, every model in here should be flashable from a PSX cart right ?
> 
Fingers crossed!

> Another question (sorry!) about those "ERASEFIRST/PROGRAM_EVERY_BYTE" flags; I'm using the PSX+cart+nops as an EEPROM programmer right now (because 😎 ) - flashing Master System 128KB and 256KB roms to SST 39SF020 to be specific - and it works ( the newly flashed EEPROM run fine in a Master System console ).
> 

I'm not sure, tbh!
There are some variations such as "ERASE_FIRST" which can only write 1s or 0's (can't remember which), so must be fully erased first. And then there are some with the "PROGRAM_EVERY_BYTE" which means "send the programming sequence before writing every single bloody byte"

I would *imagine* it does no harm just including that by default? (lmk lol?)

> Also, I'm planning on testing using a 39SF040 soon© so I should be able to give you the ID bytes for this chip.
Nice one, cheers :)

ABelliqueux, 2024-05-26:
I checked the datasheets for various chips in SST's 39 series for ID bytes and found those, maybe it would be worth it to add them in one go ? 

| Device | Address | Data | Size |
|----|----|---|----|
| Manufacturer ID | x00 | BF |  | 
| Device ID | | | |
| SST39LF/VF010 | x01 |  D5 | 128K |
| SST39LF/VF020 | x01 |  D6 | 256K | 
| SST39LF/VF040 |  x01 | D7 | 512K |
| SST39SF010A | x01 | B5 | 128K |
| SST39SF020A | x01 | B6 | 256K |
| SST39SF040 | x01 | B7 | 512K |

I also found a 512K, 5V chip from [Macronix (MX29F040CTI)](https://docs.rs-online.com/5d43/0900766b814ac74d.pdf) which looks like it could do the job, although more expensive.
There is also a [1024K, 3V chip from SST, the SST49LF008A](https://docs.rs-online.com/2773/0900766b80f4b395.pdf) that looks like it might be used, but it might be a bit more involved as it has two interface modes (Firmware Hub and Parallel Programming) and needs 3.0-3.6V. 

Here are their IDs just in case : 

| Device | Address | Man. ID | Device ID | Size | Voltage |
|----|----|---|----|----|----|
|Macronix MX29F040C | X00, X01 | C2 | A4 | 512K | 4.5-5.5V |
|SST SST49LF008A | X00, X01 | BF | 5A | 1024K | 3.0-3.6V |

JonathanDotCel, 2024-05-27:
> I checked the datasheets for various chips in SST's 39 series for ID bytes and found those, maybe it would be worth it to add them in one go ?
> 
> Device	Address	Data	Size
> Manufacturer ID	x00	BF	
> Device ID			
> SST39LF/VF010	x01	D5	128K
> SST39LF/VF020	x01	D6	256K
> SST39LF/VF040	x01	D7	512K
> SST39SF010A	x01	B5	128K
> SST39SF020A	x01	B6	256K
> SST39SF040	x01	B7	512K
> In the LF/VF datasheets, it's stated :
> 
> > The SST39LF010, SST39LF020, SST39LF040 and SST39VF010, SST39VF020, SST39VF040 are 128K x8, 256K x8 and **5124K** x8
> 
> Surely, that '5124K' is a typo and they mean 512K ?
> 
> I also found a 512K, 5V chip from [Macronix (MX29F040CTI)](https://docs.rs-online.com/5d43/0900766b814ac74d.pdf) which looks like it could do the job, although more expensive. There is also a [1024K, 3V chip from SST, the SST49LF008A](https://docs.rs-online.com/2773/0900766b80f4b395.pdf) that looks like it might be used, but it might be a bit more involved as it has two interface modes (Firmware Hub and Parallel Programming) and needs 3.0-3.6V.
> 
> Here are their IDs just in case :
> 
> Device	Address	Man. ID	Device ID	Size	Voltage
> Macronix MX29F040C	X00, X01	C2	A4	512K	4.5-5.5V
> SST SST49LF008A	X00, X01	BF	5A	1024K	3.0-3.6V

Cheers, genuinely appreciate the effort, but I'm wary about adding chips that haven't been explicitly tested.
E.g. for bullshit like the "erase first" or "program every byte" or "extended ID sequence" style bullshit.
Which have you tested?
I could probs just order a few of the rest and try them out!

ABelliqueux, 2024-05-27:
>  Which have you tested?

None yet I'm afraid. I'll be able to get some 39SF040 next month, but the other types I can get my hands on are either already supported, either too expensive (have to buy in bulk, which I'm not inclined to do just to get one). I have yet to find a seller that'll allow me to buy one of each without shipping costs to the EU that goes through the roof (mouser, farnell, rs, reichelt).

NDR008, 2024-05-27:
> <JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1/issues/37/2132837628
> @github.com>

Schnappy, what is the minimum bulk amount?

NDR008, 2024-05-27:
> It depends on the model, but it's 5 for most (rs-online).

I still owe you something for the MCs
Maybe I can order that for you.

JonathanDotCel, 2024-05-27:
> > Which have you tested?
> 
> None yet I'm afraid. I'll be able to get some 39SF040 next month, but the other types I can get my hands on are either already supported, either too expensive (have to buy in bulk, which I'm not inclined to do just to get one). I have yet to find a seller that'll allow me to buy one of each without shipping costs to the EU that goes through the roof (mouser, farnell, rs, reichelt, digikey...).

Cool, fair enough
I'll have a poke around as well, see if I can find any small batches at decent prices!

ABelliqueux, 2024-05-28:
> Maybe I can order that for you.
Hey! Cheers for that. There's something I didn't think about that Dan said; In order to  be able to flash a 512K rom to a 512K EEPROM (SST39SF040) you might need specific HW (Xplorer) because other carts mapper won't be able address the upper 256K ? 

As I only have carts with 128K/256K chips, and I suspect mappers, I'm afraid I won't be able to test right now...

JonathanDotCel, 2024-05-28:
That's right, yeah
On XFX anyway, only 384k are addressable, and I suspect the final 128k is used for the virtual memory cards.
Action replay V3 (the ribbed one which looks like a comma shape) appears to be a similar setup