JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #38: File explorer and install doen't work on Unirom 8.0.K with Duckstation.

kotenok2000, 2024-05-29:
8.0.j works  
I was even able install  and boot it from memory card with BIOS 37157331.
 8.0.K is stuck on this: 
[image]

I get tons of this in a log.
```
[ 2574.0981] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.1401] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.1423] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
[ 2574.1565] W(ProcessDataSector): Interrupt not processed in time, missed 3 sectors
[ 2574.1753] W(ProcessDataSector): Interrupt not processed in time, missed 4 sectors
[ 2574.1890] W(ProcessDataSector): Interrupt not processed in time, missed 5 sectors
[ 2574.2034] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.2380] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.2532] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
[ 2574.2693] W(ProcessDataSector): Interrupt not processed in time, missed 3 sectors
[ 2574.2710] W(ProcessDataSector): Interrupt not processed in time, missed 4 sectors
[ 2574.2878] W(ProcessDataSector): Interrupt not processed in time, missed 5 sectors
[ 2574.3037] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.3372] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.3567] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
[ 2574.3721] W(ProcessDataSector): Interrupt not processed in time, missed 3 sectors
[ 2574.3872] W(ProcessDataSector): Interrupt not processed in time, missed 4 sectors
[ 2574.4031] W(ProcessDataSector): Interrupt not processed in time, missed 5 sectors
[ 2574.4048] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.4531] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.4690] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
[ 2574.4712] W(ProcessDataSector): Interrupt not processed in time, missed 3 sectors
[ 2574.4873] W(ProcessDataSector): Interrupt not processed in time, missed 4 sectors
[ 2574.5037] W(ProcessDataSector): Interrupt not processed in time, missed 5 sectors
[ 2574.5198] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.5544] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.5703] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
[ 2574.5864] W(ProcessDataSector): Interrupt not processed in time, missed 3 sectors
[ 2574.6028] W(ProcessDataSector): Interrupt not processed in time, missed 4 sectors
[ 2574.6050] W(ProcessDataSector): Interrupt not processed in time, missed 5 sectors
[ 2574.6211] W(ProcessDataSector): Interrupt not processed in time, missed 6 sectors
[ 2574.6716] W(ProcessDataSector): Interrupt not processed in time, missed 1 sectors
[ 2574.6770] W(ProcessDataSector): Interrupt not processed in time, missed 2 sectors
```

ramapcsx2, 2024-05-29:
Hey,
could you look for SIO (or TTY) logs in Duckstation? This would also show what Unirom prints, which would probably help more :)

kotenok2000, 2024-05-29:
[duckstation.log](https://github.com/JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1/files/15487076/duckstation.log)

stenzek, 2024-05-31:
Not a Unirom bug AFAICT. Unirom sets BFRD, clears it, then sets it again, which presumably works fine on console, but confused my CD-ROM emulation. Still not sure if it's 100% correct, but Unirom seems to work fine after https://github.com/stenzek/duckstation/commit/d01343881035e0340c8af6129a4bc186db3b727f (preview build).

PS: It'll probably break in most, if not all other emus too, that have a "data FIFO" or "DMA buffer" ;)