JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #39: Final Fantasy VIII (PAL) unable to be forced to NTSC, reverts to PAL 50Hz on boot

gerardasj, 2024-06-17:
Console Model/Region: SCPH-7502, PAL
Game Region: PAL
Boot Method: MM3 modchipped PS1 with Unirom flashed to Smart Cartridge (generic Parallel Port cheat device)
What game? - Final Fantasy VIII (Platinum, if it makes any difference)
Did you burn it from one of those horrible .ECM disks? - Legit copy
Does it work from the normal boot or disc swap? - It only boots with a cheat device in the Parallel Port due to LibCrypt protection
Is the disc clean? - Yeah, it boots, but not into forced NTSC mode. Just reverts to PAL 50Hz
Cartridge unrecognised? - No, just a specific problem with Final Fantasy VIII, probably due to the LibCrypt in the PAL copy
Take it apart and let me see the chips - I don't think it's a problem with the cheat device, just the game. I've tried to do the same PAL>NTSC with FreePSXBoot exploit installed on the memory card, but in that case I couldn't boot the game at all because of the anti-mod. Do let me know if you want to see inside, but I doubt it will solve the issue.

Side note: The game works forced to NTSC on a softmodded PS2 with GSM, but manually setting the x and y offset every time is a pain.

Niche issue, not a big deal, but something that is preferrably patched.

Cheers

ramapcsx2, 2024-06-17:
I remember playing this game from PAL Disk on my console in NTSC. It needed Import Player Lite and possibly one more thing to help it booting, not sure there... but it does work eventually, and looks and plays normal in NTSC :)

gerardasj, 2024-06-24:
Can confirm the same issue with legit PAL copy of Final Fantasy IX, with NTSC mode on GSM as a workaround (PlayStation 2). The game functions properly in 60Hz, but does not like being forced into 60Hz on a real PS1. I've even tried a LibCrypt patched ISO from online, but the video signal reverts to PAL on startup. Might be more of an issue with the game rather than the copy protection.