JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #40: Legacy of Kain: Soul Reaver (PAL) bouncing image when forced to NTSC

gerardasj, 2024-07-03:
Console Model/Region: SCPH-7502, PAL
Game Region: PAL
Boot Method: MM3 modchipped PS1 with Unirom flashed to Smart Cartridge (generic Parallel Port cheat device)
What game? - Legacy of Kain: Soul Reaver
Did you burn it from one of those horrible .ECM disks? - Legit copy
Does it work from the normal boot or disc swap? - Yes
Is the disc clean? - Yeah, it boots, but has issues in 50Hz
Cartridge unrecognised? - No, just a specific problem with Legacy of Kain: Soul Reaver. The game runs slower than NTSC in default PAL 50Hz, which is why I would like to try to make it 60Hz. Squashed image when forcing to NTSC on a PlayStation 2, so definetely the game.

Cheers

JonathanDotCel, 2024-07-03:
Hi,
is it just squashed? or when you saying "bouncing" is it flickering like crazy?

gerardasj, 2024-07-04:
> Hi, is it just squashed? or when you saying "bouncing" is it flickering like crazy?

On PlayStation 1, the image bounces from the bottom to the top of the screen, like crazy, when forced to NTSC on a PAL PlayStation.

JonathanDotCel, 2024-07-04:
Aah damn, it's kinda a "generic, might work, might not" solution, but i'll have a look.
In the mean time, have you tried "Import Player Lite"? It uses a different method to patch PAL/NTSC, could work?