JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1 #45: [8.0.K] Caetla 0.38 Cheats?

AskaLangly, 2025-01-14:
When I wrote Unirom 8.0.K over to my 256K GS cart on my 7501 console, I was expecting the ability to use cheats again. 
Seems there's no way to do so. Am I missing something here?

[image]
[image]

theretrobristolian, 2025-08-10:
You are not alone in this, I've seen other people say they don't understand how you add cheats on Caetla 0.38  and I don't either! I have got it working on Caetla 0.37 with a memory card file and that seems to work fine.

Does anyone understand how this should work?

mhmetalheart, 2025-08-11:
I use HxD editor (or any HEX editor you like), cut the caetla 0.34 from 8F, then, sew it with 8K. After that, use the memcard pro to ‘restore from memory card' it. It worked very well.

mhmetalheart, 2025-08-11:
I'm not a professional programmer. Just a cheat device fan. 
As I talk with my friend (he is a huge fan, and a great programer), we both think maybe K.COMM forget to 'open' the 'cheat code' option. Because I found the option: ‘ＰＡＲ付きゲーム起動 (Game Start with PAR/GameShark)' in the ROM. But the UI haven't this option. So, we cannot choose it any more. But just our guess. Didn't check program deeper on 0.38.

ramapcsx2, 2025-08-11:
Just chiming in here, but some very late Caetla version started to have cheats on a Memory Card, as far as I know anyway!
This might be the reason you can't find the cheats anymore: They're supposed to come from Memcard.

ottelo9, 2025-11-08:
Hi, any news on this? Has anyone here ever managed to get Caetla running with Unirom 8K?

mhmetalheart, 2025-11-17:
1. in original unirom 8K, it just have 0.35, 0.37, 0.38 caetla. I added rest of them into it. And I add some modded PAR (jap verion), X-Terminator (jap version) into it. My friends and I play Japanese games, but we tired to read Japanese. So, I translated the in game search manu for them. 
2. these are the single 128K caetla, original versions. Not caetla+unirom8 K. I just make them can add cheat codes. You can find them in Caetla option.
3. No... I just share with some friends. Not published it until now. Because I'm waiting one guy to mod 0.34: to make it can stop disc automatically, can use 256K EEPROM, and try to make it can read KARAT HIGI code discs (similar with GAMESHARK THE BIGWAVES). That's a very big project.
4. You use XPLODER, right? It can work. I confirm it works on my X-TERMINATOR PRO DX (512K) very well. Same PCB hardware.
5. You can try duckstation at first. It can let you emulator a XPLODER 512K cartridge. You can install into that 'cartridge', see 'what happened' before burn a CD-R for your PS1. : )

ottelo9, 2025-11-25:
@mhmetalheart 
Thanks for your help and the image. Yesterday I tried with duckstation and a empty .bin file (write access enabled). I tried my and your modified unirom image and restored a Xploder Cheat image to my emtpy bin file. This works and after restart the Xploder boots. But the "Install on Cheat device" method (unirom + caetla) doesnt work! The image is 100% successfully written but after a restart (without CD image loaded) I got the standard screen -> Memory Card / CD Player. So I cant test unirom+Caetla on the emulated Xploder :(

I will try your image on a real CD-R. Do you know, what happens If I choose "Boot with Caetla"? The game will boot with Caetla, yes, but which cheats are activated ?

mistamontiel, 2026-09-19:
Dawh damn, someone vibing with store+load from card o.0 Next update may offer .34 or .38 pleeease